mytest4u/BMS #5: BMS SW:  DR01_16S100JC03_V2 0 0 (Victron Version)

mytest4u, 2023-01-31:
[DR01_16S100JC03_V2 0 0_T1_V.zip](https://github.com/mytest4u/BMS/files/10551424/DR01_16S100JC03_V2.0.0_T1_V.zip)

Damit funktioniert auch DIP 5 und 6 nach oben für den Masterpack wie in der Anleitung für Victron beschrieben. 
Das V in der Datei steht für Victron. Es gibt auch Firmware mit G. Das ist für Growatt und wird als Pylontech erkannt.

**SW von YT User:** 
Martin Hilburger  mit Version V2.92 am Cerbo im Einsatz

mytest4u, 2023-02-01:
Anleitung (von Martin Hilburger)
1. PC und Akku mit Kabel verbinden (DIP 1 muss auf ON Stehen für PC Verbindung)
2. Anwendung „DR-Aplication.exe“ öffnen und Fenster offen lassen
3. Im Ordner „dist“ die „index.html“ im Chrome Browser öffnen.
4. Im Reiter „Help“ Sprache auf englisch stellen ggf Passwort 666666 eingeben.
5. Im Reiter „Setting“ update wählen, die Bin Datei auswählen und Flashvorgang starten.( eine grüne LED am Akku blinkt)
6. Fertig
7. Jetzt den Masterpack mit DIP 5 und 6 nach oben anschliessen wie in der Anleitung für Victron Verbindung.
8. Pack wird als „LFP battery“ erkannt unter Details sind jetzt die Spannung der höchsten und niedrigsten Zelle zu sehen.
9. Ausserdem wird angezeigt wieviel Packs online sind und ob ein Pack die Ladung/Entladung verhindert.
10. DIP 6 nach oben wird als „Pylontech“ erkannt Multiplus und Laderegler laden den Akku nicht mehr deswegen diesen Modus nicht verwenden.

SW zum lesen und hochladen:
[DR_22.09.01-5.zip](https://github.com/mytest4u/BMS/files/10557925/DR_22.09.01-5.zip)
Tool von Superbattery, Änderungen mit Passwort 666666

LeoWinterDE, 2023-02-11:
The screen of the web ui software (DR22.09.01-5) shows the following information for this firmware:
```
Product info 2.2
Hardware model T1_V
Product model DR01
Project code 16S100JC03
Software version 2.0.0
Boot version 1.1
```

Fabi268, 2023-02-14:
Hi, 

habe das Update auch gemacht, der Victron erkennt die Batterie jetzt auch als LFP. 
Wenn ich auf den Batteriemonitor gehe zeigt er mir nur die Daten von einer Batterie an auch deren Kapazität.

Batteriemodule 1 Online 0 offline 

Ist das Normal?

theimo1221, 2023-03-06:
Nabend @mytest4u,
Woher hast du die Datei?

Habe die 200A Variante (16S200) da geht diese Datei nicht.

Vielen Dank und Grüße 

Thiemo

joggl, 2023-03-26:
Funktioniert das BMS eigentlich bei irgendjemandem richtig. Mit einem Batteriepack war das noch in Ordnung, seit ich jetzt den zweiten Pack nachbestellt hab und angeschlossen ist nur noch Chaos. Der neue Batteriepack fällt immer weiter ab, die Distanz der beiden Packs wird immer größer. Teilweise Zellendifferenzen in einem Pack zwischen 3,55V und 3,33V. Das ist doch nicht normal. Habe die 120AH Version mit dem Update V1.3 aufgespielt. Super Battery Store ( oder jetzt auf einmal Pappol) sagt es würde keine neuere Software geben. Gruß Jochen

joggl, 2023-03-26:
[image]
[image]

joggl, 2023-03-26:
Nach 3 Stunden sind die einzelnen Zellen ausgeglichen, aber der SOC meilenweit auseinander! Und das wird täglich mehr...

joggl, 2023-03-26:
[image]
[image]

theimo1221, 2023-03-30:
I have a quite similiar setup:
First image is the 200 Ah Master Unit, the 2nd Image is the 230 Ah as the "1 Slave", don't worry about the different dates, as I never connected my pc to the slave to change the clock (the battery came shipped with the newest firmware installed).

[image]

[image]

Personally I'd say my system is in a nice balance, as both batteries have a almost similar cell voltage range.

But and that's interesting, **why does my main (smaller) Battery is already 50Ah down, whilst the slave (bigger Battery) is just down 20Ah**?

ThePunisher1234, 2023-03-31:
> Hallo,
> 
> bin auch auf der Suche nach der Victron Firmware für die 16S 200Ah Variante. Der SOC springt teilweise sehr stark beim Laden. Bei mir wird das BMS als Pylontech erkannt. Das funktioniert allerdings nur mit Venus-OS Version ≥ 3.00 da sonst bei Pylontech-BMS auf 15S Spannungsbereich begrenzt wird.
> 
> Ich habe die Victron-spezifische Firmware für die 200Ah-Variante erhalten und werde diese jetzt testen. Wenn alles läuft, wo kann ich die Firmware hin senden um sie hier verfügbar zu machen?
> 
> Grüße Baerbel

Genau die Firmware suche ich auch... Kannst du sie zur Verfügung stellen?

ThePunisher1234, 2023-04-03:
[wetransfer_2-upgrade-operation-video-wmv_2023-04-03_0901.zip](https://github.com/mytest4u/BMS/files/11137015/wetransfer_2-upgrade-operation-video-wmv_2023-04-03_0901.zip)
Hier die Firmware und Software inkl Video für den 200AH Akku

Scheppo, 2023-04-23:
Ich kann alles auslesen und verstellen in der Software, aber ein Update gelingt nicht.
Beim Start vom Update blinkt die vierte led 3mal und dann macht der bms einen reset und piept einmal.
Danach kann ich alles normal weiter auslesen aber ein Update hat nicht stattgefunden, der blaue Balken erscheint auch erst garnicht. Habe alle möglichen Baudraten ausprobiert.
Hat jemand einen tip?

theimo1221, 2023-04-23:
Ist das update was du ein spielst für dein Produkt?
Habe unterschiedliche update dateien für die 200ah variante als für die 230ah

Scheppo, 2023-04-23:
Ja klar, für 100ah.
Der Verkäufer hat es per Remote auch nicht geschafft es auzuspielen

theimo1221, 2023-04-23:
Pins im Modus für PC Anschluss und Verbraucher abgeklemmt?
Batterie ausreichend geladen?

Scheppo, 2023-04-23:
Dip auf 1 die anderen off
Alles abgeklemmt
Battery fast voll
Die Kommunikation mit den 115000 Baud klappt nicht, habe fast den usb Stick in verdacht.
Obwohl der Akku bei jederbaudzahl im Update einen reset macht

Scheppo, 2023-04-23:
> I have a quite similiar setup: First image is the 200 Ah Master Unit, the 2nd Image is the 230 Ah as the "1 Slave", don't worry about the different dates, as I never connected my pc to the slave to [...]

Das ist ja normal, um es leichter zu erklären nehmen wir an du hast 2 Batterien parallel geschaltet.
Eine hat 200ah und die andere 100ah.
Wenn du jetzt mit 30 Ampere laden würdest dann würde die 200ah Batterie 20A und die 100er nur 10A nehmen.
Das selbe passiert auch beim entladen.
Ich habe eine 280ah und 120ah parallel und da ist das Verhältnis beim Laden und entladen ca. 2.5

theimo1221, 2023-04-23:
Sorry, habe mich vielleicht falsch ausgedrückt. Meine nicht die Lade-/Entladeströme sondern die Mengen.
Warum ist die 200ah Batterie schon auf 150ah (25% entladen) während die andere nur 21ah entladen ist sprich -10%.

Scheppo, 2023-04-23:
> Sorry, habe mich vielleicht falsch ausgedrückt. Meine nicht die Lade-/Entladeströme sondern die Mengen.
> Warum ist die 200ah Batterie schon auf 150ah (25% entladen) während die andere von 230ah nur 21ah auf 209ah entladen ist sprich -10%.

Ok, Messe mal den Innenwiderstand von beiden Akkus an den Anschlüssen außen, wenn der zuviel Unterschied hat kann es daran liegen. Die internen BMS sind auch nicht genau, alles unter 0,6-1 A wird nicht erfasst und somit nicht gemessen.

dingslebumsle, 2023-04-23:
> Ist das update was du ein spielst für dein Produkt? Habe unterschiedliche update dateien für die 200ah variante als für die 230ah

Könntest du die Firmeware für 230Ah auch bitte zur Verfügung stellen. Das wäre sehr freundlich.

MfG

ThePunisher1234, 2023-04-23:
> Das ist ja normal, um es leichter zu erklären nehmen wir an du hast 2 Batterien parallel geschaltet. Eine hat 200ah und die andere 100ah. Wenn du jetzt mit 30 Ampere laden würdest dann würde die [...]

Das ist alles völlig richtig, ich habe zwei 200AH Batterien Parallel, eine zeigt noch 3% und die andere 15%, noch keine Ahnung wie ich das über das BMS glatt gebügelt bekomme. wohlgemerkt Baugleiche und zum gleichen Zeitpunkt gekauft und in Betrieb genommen.

Hat sich mal jemand mit der Kalibrierung des BMS beschäftigt? das eine zeigt eine Spannung von 52,12V das andere 52,17V

joggl, 2023-04-24:
Die SOC Werte meiner 2 Pack's laufen auch immer mit der Zeit auseinander, die Zellspannungen der einzelnen Pack's sind aber immer fast identisch. Ich habe die Ladespannung im DVCC auf 55,2Volt eingestellt. Wenn die SOC der Batterien zu weit auseinander gelaufen sind erhöhe ich die Spannung im DVCC schrittweise bis auf 55,8V. Wenn die einzelnen Zellspannungen dann eine bestimmte Spannung erreichen "Nullt" sich das BMS wieder und zeigt dann dass die Batterie voll ist, also SOC 100%. Hat der Typ von der Offgrid Garage mal in einem Video über ein anderes BMS erklärt, ist hier auch so.
Gruß Jochen

ThePunisher1234, 2023-04-24:
> Die SOC Werte meiner 2 Pack's laufen auch immer mit der Zeit auseinander, die Zellspannungen der einzelnen Pack's sind aber immer fast identisch. Ich habe die Ladespannung im DVCC auf 55,2Volt [...]

Ich erreiche mit 55,2Volt auch die 100%, die Akkupacks sind von der Spannung natürlich identisch, die BMS`zeigen unterschiedliche Spannungen im Display, Differenz um 0,05V. 
Der SOC driftet bei mir ab 15% nach unten ab. Gestern erst gesehen, ein Akku zeigt 13% der andere ist schlagartig bei 3%
Die Zellspannungen, sowie die Gesamtspannungen der Akkus sind natürlich identisch. Wir bewegen uns da mit allen Zellen bei 3,09V also alles noch unkritisch. Das Problem mit dem SOC ist, ich würde den Akku gerne bis 10% entleeren, da aber ein Akku schlagartig auf 3% fällt und der andere bei 13% ist, macht das Victron System aus beiden Werten mal schnell 8% und fängt dann wieder das laden auf 10% an.

Den Zelldrift bekommt man super glattgebügelt mit einem einfachen Top-Balancing auf 3,65V und einem aktiven Balancer. Hab ich genau so einmal gemacht, Active Balancer eingebaut und mit externen Netzteil auf 58,4Volt geladen. Seitdem sind alle Zellen super nah beieinander.

joggl, 2023-04-24:
> Ich denke es hat auch hiermit zu tun: Ab 90% SoC beginnen die Min/Max Spannungen der Zellen auseinander zu laufen. [image]

Meine Min/Max Zellspannungskurve sieht genauso aus, ob das noch OK ist weiß ich nicht...
[image]

ThePunisher1234, 2023-04-24:
[image]
Bei mir nicht, 0,03V Differenz

joggl, 2023-04-24:
> [image] Bei mir nicht, 0,03V Differenz
> 
> Wie gesagt, habe ein gutes Top-Balancing gemacht

Das sieht wesentlich besser aus!
Was braucht man um das Top Balancing durchzuführen?

ThePunisher1234, 2023-04-24:
> Das sieht wesentlich besser aus! Was braucht man um das Top Balancing durchzuführen?

Habe mir einen aktiven Balancer gekauft, den angeschlossen und über ein Netzteil geladen auf die 58,4Volt.
Klar du musst den Deckel des Akkus öffnen, die einzelnen Balancer-Leitungen durchtrennen und dann den neuen Balancer dazwischen anlöten. Ist kein großes Hexenwerk, etwas nervig vielleicht. 

Habe den Balancer hier genommen: [ https://www.amazon.de/dp/B09RWZ1VGY?psc=1&ref=ppx_yo2ov_dt_b_product_details](https://www.amazon.de/dp/B09RWZ1VGY?psc=1&ref=ppx_yo2ov_dt_b_product_details)

Andy beschreibt es hier ganz gut [https://www.youtube.com/watch?v=ZplbAAqnrjo](url)

theimo1221, 2023-04-26:
I ordered the active balanced presented by Andy in a different video: 4TH NEEY 4A Smart Aktive Balancer 3S 4S 5S 6S 7S 8S 14S 16S 20S 21S 22S 24S Lifepo4 / Lipo/ LTO Batterie Energie ausgleich
https://a.aliexpress.com/_EJ3iHzp

This one can be used without cutting the cables or soldering as you can just take the plugged cell cables from the bms.

I'll let you know about the results once I receive the product.

ThePunisher1234, 2023-04-26:
> I ordered the active balanced presented by Andy in a different video: 4TH NEEY 4A Smart Aktive Balancer 3S 4S 5S 6S 7S 8S 14S 16S 20S 21S 22S 24S Lifepo4 / Lipo/ LTO Batterie Energie ausgleich [...]

Bekommt dann das BMS noch die Werte der Zellen?

theimo1221, 2023-04-27:
Ich werde den Balancer nicht dauerhaft anschließen, sondern eben nur bei Bedarf verwenden.
So kann ich ihn auch für beide Packst nutzen.

Scheppo, 2023-05-10:
Weiß jemand wie man ein kaputtgeflashtes bms wieder zum Leben erweckt?
Es hing beim Update in einer Schleife und ging nicht bis zum Ende durch.
Nun wird es nicht mehr erkannt.

mvdh02, 2023-05-22:
Könnte jemand einen Screenshot der Einstellungen posten? Wie Sie diese mit der Software sehen. Und ich habe die Vorstellung, dass die Einstellung im BMS bei jeder Batterie unterschiedlich ist. Danke

kopierschnitte, 2023-07-19:
> Weiß jemand wie man ein kaputtgeflashtes bms wieder zum Leben erweckt? Es hing beim Update in einer Schleife und ging nicht bis zum Ende durch. Nun wird es nicht mehr erkannt.

Selbes Problem hier, jedoch mit offizieller Firmware vom Händler. Bin jetzt sehr gefrustet :-(

ThePunisher1234, 2023-07-25:
> 

Könntest du die Firmware zur Verfügung stellen?

kopierschnitte, 2023-07-25:
Habe ich im Fork als Pull-Request: https://github.com/christian1980nrw/BMS/pull/1

FloFry, 2023-08-16:
> "Das V in der Datei steht für Victron. Es gibt auch Firmware mit G" --> Wo kann ich denn die Firmware mit dem "G" für Growatt finden?

Vielen Dank!

dingslebumsle, 2023-08-16:
> Okay, der Händler hat's für mich remote gefixt. Es gibt wohl eine Version "DR06_16S200JC03_V2.0.0_**T2**_V_**R**" und außerdem gibt es für solche Update-Schleifen einen speziellen Bootloader-Code.

Ist diese Firmware auch für die 230AH Version, bzw. gibt es die auch ohne Victron? Werd noch wahnsinnig mit dem BMS :(.

kopierschnitte, 2023-08-16:
Also die Dateien, die ich oben (im Link) hochgeladen habe sind definitiv für die 230AH Version. Hab aber nur die mit V bekommen... sorry!

Ja, das BMS ist wirklich zickig.

Hilfii, 2025-01-14:
> aber

Hast du das Netzteil direkt an die Zellen angeschlossen oder über die Kontakte am Gehäuse(quasi mit BMS)